Levi--G/USBLibrarySTM32 #2: Using Joystick Libary with SerialUSB Communication

cagri-yuksel, 2022-10-18:
Thank you for hard working, i tried to use SerialUsb method but device didn't recognize by windows. 
Also tried to use only FFB libary but i got some error. 
I am making mistake or not yet supporting those features?

Levi--G, 2022-10-19:
Did you follow the installation instructions? Can you send me your code for testing? The supported libraries are in the readme. FFB is not one of them yet, i didn't even know it existed ;) if you can send me some code that works on an avr chip i can take a look to make it work on stm32. :)

Levi--G, 2025-05-03:
Closing since the issue is too old, if this is still an issue after the 3.0 update feel free to reopen.

closed 2025-05-03 by Levi--G