Levi--G/USBLibrarySTM32 #4: Custom pluggable SerialUSB library?

elMariachi90, 2024-02-14:
First of all thank you for this amazing library which is allowing me to program my Blue Pills as game peripherals through Platformio. Can't say how much I apreciate it.

I am here to ask if it is possible in any way to add the Serial over USB functionality on top of the Game Peripheral HID.
I really need my application to send some Serial messages through USB, but I can't find any reference to the "Custom Pluggable SerialUSB library" you mentioned in your description.

Can you please help me getting this to work?

Levi--G, 2024-02-14:
Hi, thank you for the kind words. I was about to go to bed but if I'm not mistaken there should be an example in the repo with the serial usb.
It should be along the lines of defining 
SerialUSB_ SerialUSB;
in your main and using that as your serial object. If you can't find it leave me a message and I'll respond first thing tomorrow :)

Levi--G, 2024-02-14:
The example is here https://github.com/Levi--G/USBLibrarySTM32/blob/main/examples%2FSerialUSB%2Fsrc%2Fmain.cpp

elMariachi90, 2024-02-15:
Wow, I didn't expect you to write back so fast.. can't thank you enough for your help!

I succeeded implementing SerialUSB + Joystick with custom HID on a chinese generic STM32F103C8 through Platformio.

Following and blending the two examples presented on this library was successfull!

Levi--G, 2024-02-16:
Yes I will take a look tomorrow, in the mean time can you send me your code tou use, or a minimum example that would really help :)

Levi--G, 2024-02-17:
Hi, I have taken a look at this but i see no issue, yes the serial will get flushed, but that is at usb level, your computer should still handle the cdc messages even if no program listens to the terminal port. Unless your terminal actually closes the usb device rather than the cdc port, or your code is stuck somewhere i see no reason why it would stop working when you disconnect CDC and i don't see this on my end.
So if you could send me the relevant bits of your code I can take a look, but i see no issue at the moment on my end.
The USB_Running will always return true for as long as you don't remove the usb cable that is normal, and SerialUSB should only return true when you have a terminal attached, but this can be platform/terminal dependant, it could very well be that will not return false once connected, but better like this than returning false when it is connected ;)

numanair, 2024-02-17:
Without seeing your code it's just a guess, but might you have a 'while' statement somewhere blocking things?

elMariachi90, 2024-02-17:
Hello guys, I didn't know it was possible to share here, but anyway here is my Platformio project.
It really seems that the SerialUSB.print hangs things up unfortunately.

Thank you guys in advance.

[Custom Handbrake Levi.zip](https://github.com/Levi--G/USBLibrarySTM32/files/14319739/Custom.Handbrake.Levi.zip)

Levi--G, 2024-02-18:
@elMariachi90 I see what you did, can you try running without the SerialUSB.flush(); after your println. That will instruct the stm to wait until it knows the PC got the message. Does that improve?

elMariachi90, 2024-02-18:
I have tried with and without SerialUSB.flush();
I have tried with SerialUSB.print(), SerialUSB.write()... any possible combination, but they all hang up the gaming peripheral functionality unfortunately..

I believe that any sort of SerialUSB call hangs the MCU is a frozen state when the serial port gets closed.. could you give it a try and let me know?

Levi--G, 2024-02-26:
No sorry im very busy this week, if possible could you make a minimal POC without external dependencies and how i can reproduce it? there is quite a lot of code in your project that has nothing to do with the communication, it would make it a lot easier for me. :) Also was this on a legit chip or a clone?

Levi--G, 2024-03-01:
Ok, i have been able to reproduce the issue, however, your project is not using the standard setup, i suggest you also redo the installation of the library and follow each step closely, as your project uses the wrong defines to enable usb, which might in the long run cause different issues. :)

Levi--G, 2025-05-03:
Version 4.0 will have configurable RX and TX buffers that will hopefully fix all freezing issues

Levi--G, 2025-05-10:
This should be fixed in 1.0.0 release, if you still have issues you can reopen the issue.

closed 2025-05-10 by Levi--G