Levi--G/USBLibrarySTM32 #5: STM32F401 Black Pill  SerialUSB + Joystick

elMariachi90, 2024-02-15:
Hello there, still me.

After succeeding implementing both SerialUSB and custom HID with Joystick library on a F103, I am here to ask if it is possible in any way for you to help me unlocking this possibility also on the better performing and more reliable F401 Black Pill.

Let me attach some info here..
I am running your library on the latest Platformio release, and this is the .ini file config I'm attempting to use.

```
`[env:genericSTM32F401CD]
platform = ststm32
board = genericSTM32F401CD
framework = arduino

build_flags = 
    ;-D USBD_REENUM_DISABLED
    -D USBCON
    -D PLUGGABLE_USB_ENABLED
    -D USBD_USE_HID_COMPOSITE
    -D USBD_VID=0x0CD4
    -D USBD_PID=0x1010
    '-D USB_PRODUCT_STRING="Custom Handbrake"'

lib_deps = 
    levi--g/USBLibrarySTM32@^0.2.0
    mheironimus/Joystick@^2.1.1
    thomasfredericks/Bounce2@^2.72`
```
    
   
But attempting to compile the same code which works flawlessy on a Blue pill, the compiler gives me the following attached errors..
    
Is there anything I can do to solve this? Or could you help in any way?
    
    
Thank you so much in advance, your support has already proven to be unvaluable!
    
[image]

Levi--G, 2024-02-17:
Hi again :)

Unfortunately the 401/411 chips have an entirely different USB implementation, and i have not been able to get it working myself. At the moment only USB FS device chips are supported, the USB OTG (like 401/411 and other newer chips) don't work. I have spent many weekends on this combing through the stm hal and middlewares without success.

The error you are seeing should not occur however, are you sure you are running the latest version?

Levi--G, 2024-02-17:
Additionally i don't know how much sense it even makes to make this library compatible for 401/411 series. this library makes it so you can implement multiple devices at the same time, but the 401/411 only support 3 endpoints as opposed to the 8 from the 103 series, so if you use 2 for the CDC you barely have enough to run smth else with only one endpoint...
So for now im making that a lower priority, if i have time ill first make it so 103 can be more easily used.

elMariachi90, 2024-02-17:
Okay, thank you for your explenation.

I have a friend that is regularly using F401s with his company and he is able to get Serial, USB FS and everything going even on F401. Unfortunately he is using a commercial solution and in my case I would just like to find something available for me to program with the Arduino framework.. 

So you definitely think there is nothing available or any possibility to implement SerialUSB + Gaming peripheral on a F401 black pill in any case? 

I know 103 is more popular, but many of them come with fake chips from China and they're not reliable at all, it seems to me that 401 is a more robust solution also because there aren't any fake boards around.

Thank you for your support!

Levi--G, 2024-02-18:
> I have a friend that is regularly using F401s with his company and he is able to get Serial, USB FS and everything going even on F401. Unfortunately he is using a commercial solution and in my case I would just like to find something available for me to program with the Arduino framework..

Can you ask him if that is on the same controller or different ones, because according to the STM datasheet there are only 3 free endpoints, if you use 2 for CDC (serial) you can only use 1 more for mouse OR joystick, and not smth like feedback joystick for example which also uses 2.

> So you definitely think there is nothing available or any possibility to implement SerialUSB + Gaming peripheral on a F401 black pill in any case?

There is, if you are willing to study the stm32 HAL and cubeMX which are the "normal" stm32 development tools. Arduino is more for hobby usage and led blink usage, where the HAL and cube are much more difficult but also more powerful, this library is written with the HAL and cube

> I know 103 is more popular, but many of them come with fake chips from China and they're not reliable at all, it seems to me that 401 is a more robust solution also because there aren't any fake boards around.

Actually the clones i got, mainly GD ones and a few CK, all work perfectly except for the flash, and i even test this library and usb on those to make sure its compatible, since i know most will end up with clones. And thus far i noticed no difference between clones and genuine stm regarding usb

and actually regarding that topic, GD is actually making f4xx clones as well, so either way its a lottery if you don't use official channels to buy your chips. (and solder them yourself because you won't find a board with them)

> I have found this other framework which unfortunately seems to be deprecated, but it seems it got the usb stuff working (?)
Maybe it's worth a check?

That is the "maple core", tbh i would really recommend against it, it does a lot different and was only retrofitted to supply "some" f4 support but that was right before it was abandoned. A lot might not work. Most libraries won't even support it.

Ill see after i reworked the library a bit so integrate better if i can support it.

Levi--G, 2024-02-18:
I added a new branch for F4 compatibility and did some work i knew needed to be done but sadly this is where i will need to stop. The code compiles and runs fine from the perspective from the controller, but computers won't actually receive any data. troubleshooting this i would need to get a logic analyser i don't have atm (my current one is too slow for usb). If anyone wants to pick it up feel free to do so.

Levi--G, 2024-02-19:
> I would, but unfortunately I would need to start from 0 as this is not really my field, I have 0 experience with this low level code.
Don't worry i don''t blame you, a few years ago i didn't either, and if i knew how much of a hassle it would have been i would have never begun this library 😆 
> The nightmares I've had has been with WCH F103 Mcus..
Ah i havent gotten those yet, those arent even cortex cores if i remember correctly, they arent even clones, just misbranded completely different MCU's xD
> I have a quite decent logic analyser as I am an electronic engineer
I might take you up on that, since im mostly a programmer and only hobby/subMhz pcb/digital engineer none of my equipment can measure the 12MHz usb signal to troubleshoot communications. Ill see when i can work some more on it and provide you with a few tests to capture :)

elMariachi90, 2024-02-19:
> Ah i havent gotten those yet, those arent even cortex cores if i remember correctly, they arent even clones, just misbranded completely different MCU's xD
>>
Yeah but unfortunately when ordering from China you never know what you're going to receive...

> I might take you up on that, since im mostly a programmer and only hobby/subMhz pcb/digital engineer none of my equipment can measure the 12MHz usb signal to troubleshoot communications. Ill see when i can work some more on it and provide you with a few tests to capture :)
>>
Okay, sounds super cool! Let me know what to test as soon as possible, I would love giving this a go! Is there any way to get in touch other than comments here which would make testing easier maybe?

Levi--G, 2024-02-21:
> Is there any way to get in touch other than comments here which would make testing easier maybe?
I have a discord that i don't really use anymore, but you can join there i guess, its where i posted my old project updates: https://discord.gg/<delete this>vs9CdTD

Levi--G, 2024-03-02:
401 is just after setup phase never replying to pc, causing it to reset the connection, this could be a txbuffer issue, an IRQ issue, a config issue or anything else really. It is very hard to debug this

Levi--G, 2025-05-10:
This should be fixed in 1.0.0 release, if you still have issues you can reopen the issue.

closed 2025-05-10 by Levi--G