Levi--G/USBLibrarySTM32 #6: Instruction to use with Arduino IDE?

lucadentella, 2025-03-11:
Hi

thanks for this library! I'd like to use it - if possible - with Arduino IDE (I'm using IDE 2.x).
The library is not available in the library manager, I downloaded the repo as a ZIP file but when I try to manually add it ("Add .ZIP library") I get the following error:

[image]

Moreover, I'm using the latest (2.9.0) version of the official (stm32duino) core, should I replace the header files in:

`AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.9.0\libraries\USBDevice\src`

?
Thanks!

Levi--G, 2025-04-28:
Hi, sorry for the delayed response, I'm first still working on getting it compatible with the newer stm arduino core on platformio/pioarduino then arduino ide support will follow after that.

Levi--G, 2025-05-03:
The library cannot work on arduino until they fix their library properties, more info here:
https://github.com/arduino/arduino-cli/issues/2908

blekenbleu, 2025-08-12:
Using Arduino IDE v2.3.6 to build MIDIUSB_loop.ino from MIDIUSB Library for Arduino v1.0.5
In IDE Tools, setting USB support "CDC (generic 'Serial' supersede U(S)ART)"
 defines `USBCON` and `HAL_PCD_MODULE_ENABLED` for builds.

Then, in `MIDIUSB/src/MIDIUSB.h`,
changed:   `#if defined(ARDUINO_ARCH_AVR)`
... to:  `#if defined(ARDUINO_ARCH_AVR) || defined(HAL_PCD_MODULE_ENABLED) `

.. then build failed for not finding files in `USBLibrarySTM32/include/`
After moving those files `USBLibrarySTM32/`,  then build continued until:
```
PWM_FullConfiguration.ino.elf section `.rodata' will not fit in region `FLASH'
arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 572 bytes
```
This is for Blue Pill target with HID Bootloader 2.2 upload method;
I guess that sketch build would have included both your USB library and STM32duino's.

Levi--G, 2025-08-13:
Arduino IDE is still not officially supported since the tools option has no option to use external USB libraries, if you select UART in the tools it will use the internal one of the stm32 platform, so don't select that. The builtin library does not have an USBAPI.h that is compatible with arduino libraries, so the first error is normal. The multiple definitions are because the builtin USB is also included and they use the same STM32 backend to provide the usb functionality.

fpistm got it working like this:
Don't select usb support in tools
Clone the https://github.com/Levi--G/USBLibrarySTM32 repository to the Arduino library folder. (note: at this step the library was not recognized)
Create an empty USBLibrarySTM32.h in the src folder and add #include "USBLibrarySTM32.h" then the library is properly recognize by the IDE
Add the path to the include folder in the compiler.stm.extra_include thanks to the platform.local.txt.
Add a build_opt.h file with: -DUSBCON -DHAL_PCD_MODULE_ENABLED

blekenbleu, 2025-08-16:
In the `NicoHoodHid` example, from where is `HID-Project.h` to be sourced?

Levi--G, 2025-08-16:
Did you delete a reply? Cause i got a notification but cannot find it?

blekenbleu, 2025-08-16:
Sorry!
Yes, the NicoHoodHid example build wanted sorting
- installing the HID-project library
- adding `#include "USBCDC.h" `

... but decided instead of  piecemeal updates
to wait and document everything wanted to compile and run MIDI on a Black Pill...
FWIW, not knowing where else `ARDUINO_ARCH_AVR` gets used,
instead added `-DUSBLIBSTM32` and, in HID-project and MIDIUSB libraries:
`#if defined(ARDUINO_ARCH_AVR) || defined(USBLIBSTM32)`

I am also trying to add a core library USBSerial, to enable
- USB support "CDC (generic 'Serial' supersede U(S)ART)"

Also FWIW, Arduino 2 does not run for me,
except when invoked from an MSYS64 bash terminal,
because my Windows 11 runs a RAM disk,
so am instead experimenting with VSCode-Arduino:
https://blekenbleu.github.io/static/VSCodeArduino/

blekenbleu, 2025-09-05:
I got code to compile and load to a Black Pill from VSCode-Arduino. then Arduino 2:
- moved the `USBDevice` folder out of `Arduino15/packages/STMicroelectronics/hardware/stm32/2.11.0/libraries`
   - Arduino 2 scripting wanted to use `USBDevice` source, whether referenced or not..
- added `USBLibrarySTM32` to `Documents/Arduino/libraries`
- in `USBLibrarySTM32`, moved files from `include` to `src`
- in `SerialUSB` example, moved `src/main.cpp` to `SerialUSB.ino`, added `build_opt.h`
   - added LED blink code to `SerialUSB.ino`, for deducing execution state
Windows 11 detects the Black Pill as COM8, but code gets stuck in `while (USBSerial) {}`
- Arduino Serial Monitor or PuTTY can connect, but no data passes...

[SerialUSB.ino.txt](https://github.com/user-attachments/files/22181667/SerialUSB.ino.txt)

blekenbleu, 2025-09-05:
Here is Black Pill COM8 data from USB Tree Viewer https://www.uwe-sieber.de/usbtreeview_e.html

[BlackPillCOM8.txt](https://github.com/user-attachments/files/22181696/BlackPillCOM8.txt)