Levi--G/USBLibrarySTM32 #7: getting leds status for keyboard hid

shoei1944, 2025-05-06:
hi
how to get leds status for keyboard hid?

i use stm32f103 bluepill and stm32duino on platformio
using a standard arduino keyboard lib

`lib_deps = 
	https://github.com/nRF24/RF24
	https://github.com/Levi--G/USBLibrarySTM32
	https://github.com/arduino-libraries/Keyboard
	https://github.com/stm32duino/STM32RTC
	https://github.com/stm32duino/STM32LowPower
	https://github.com/FastLED/FastLED`

Levi--G, 2025-05-06:
Hi,

The keyboard library you are using does not support the keyboard leds, you might need to find a different library or ask the library creators to add the functionality as this is not something that my USB implementation provides.

I hope that answers your question?

Levi--G, 2025-05-06:
This one for example does support leds: https://github.com/NicoHood/HID/blob/master/src/SingleReport/BootKeyboard.cpp

shoei1944, 2025-05-07:
with [https://github.com/NicoHood/HID](https://github.com/NicoHood/HID)
i got so many errors
[pastebin](https://pastebin.com/s0LmdYzw)

do you know other libraries for working with keyboard and leds support?

Levi--G, 2025-05-07:
You can use NicoHood, you just need to add -D ARDUINO_ARCH_AVR to your defines since they havent added STM32 support. but since this library is compatible with AVR implementations that should work. I will create an issue over at NicoHood to add stm32 support and create an example in the mean time.

Levi--G, 2025-05-07:
> I checked this library, it really works, but VERY slowly, the polling frequency barely exceeds 65~Hz, but with the Arduino library the polling frequency is 1000Hz and the keys do not "lag/stick"
> is it possible to somehow implement getting the caps lock indicator in another way?
> 
> edit: after experimenting I found out that the possible cause of the performance problems was the module nrf24l01

Yea the leds check is literally getting a variable from ram,  that cannot cause delay xD glad you found it. Is there any other issue?

shoei1944, 2025-05-07:
slightly unrelated to this question but..

just 1 question, is there an implementation of webusb?

Levi--G, 2025-05-08:
> slightly unrelated to this question but..
> 
> just 1 question, is there an implementation of webusb?

Im not sure what you mean, webusb is used in a browser on the host side, this is a client side library and the stm cannot run javascript?

shoei1944, 2025-05-08:
like this https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/WebUSB/webusb_serial/webusb_serial.ino

Levi--G, 2025-05-08:
Okay so webusb really isnt a usb standard but rather a browser just checking every connected device, this has little to do with the usb connection and the avr api support, you could just rewrite the part you need from the adafruit library to a pluggableusb module and then load that. But that is not something i will add in the base library. Either way its redundant cause you can just use a normal usb device on the web either way by coupling it as a serial/other device with the brower apis. ESP web tools for instance does that with their flash tool, they can connect to any device you select in the browser and upload new firmware to it.

closed 2025-05-09 by shoei1944