Levi--G/USBLibrarySTM32 #8: errors after update lib

shoei1944, 2025-05-11:
hi again
i get some errors after update ur lib

https://pastebin.com/gXsXG1uk

platformio.ini:
```
[env:genericSTM32F103CB]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = hid
build_flags =
 	-D ARDUINO_ARCH_AVR 
	-D USBCON
	-D HAL_PCD_MODULE_ENABLED 
lib_deps = 
	https://github.com/nRF24/RF24
	https://github.com/Levi--G/USBLibrarySTM32
	https://github.com/NicoHood/HID
	https://github.com/FastLED/FastLED
	https://github.com/stm32duino/STM32LowPower
	https://github.com/stm32duino/STM32RTC
```
	

includes:
```
#include <Arduino.h>
#include <SPI.h>
#include <RF24.h>
#include "HID-Project.h"
#include "USBCDC.h"
#include "layers.h"
#include <FastLED.h>
#include <EEPROM.h>
#include "STM32LowPower.h"

```

Levi--G, 2025-05-11:
I will take a look tomorrow, I think I forgot to remove a file in the release version. Can you try deleting that src/usbd_ep_conf.c file and see if that fixes it?

Levi--G, 2025-05-12:
Ill get it fixed in 1.0.1 that file should have been deleted but somehow isn't. My bad :)