Levi--G/USBLibrarySTM32 #11: problematic USBSerial.availableForWrite()?

blekenbleu, 2025-09-07:
## Description of Issue
After establishing a terminal session between Windows and Black Pill,
then closing the Windows terminal app
- if the sketch continues writing, then sketch data will no longer be returned
   when terminal app eventually restarts.
- I suppose this is because sketch write data eventually overflows the available buffer,
  but `USBSerial.availableForWrite()` continues returning more than 25.
- current workaround in the sketch is to pause writing after 10 seconds with no data received

## Technical Details
- Arduino Board (STM32 Black Pill): 
- Host OS (Windows 11):
- Arduino IDE 2.3.6

## Sketch to reproduce  // comment out line 97 to provoke the bug
~~~
https://blekenbleu.github.io/static/USBLibrarySTM32/SerialUSB.ino
~~~

## Setup Details
A10-Gnd jumper to enable DFU (known OG Black Pill bug)

## Additional info (optional)
Is there some other technique for sketches to discover whether COM port sessions are (no longer) active?

Levi--G, 2025-09-07:
I will take a better look tomorrow but in the mean time, are you sure you are not writing more **bytes** than you are checking for including special characters like /r /n etc? You are not writing unicode or anything by chance?

There are options specifically to prevent wait hangs in the options header as well you can check out, you can define them globally to overwrite them. The defaults prevent data loss by hanging until flushed but if you do not require data integrity you can switch to an overwrite approach. The available to write should be accurate though. You are using the latest version correct?

blekenbleu, 2025-09-07:
Testing for 25 available, sending 20 (or 21 if string-terminating 0 gets sent)
```
  if (millis() > 10000 + now)   // suspiciously quiet
  {
    now = millis();
    if (25 < USBSerial.availableForWrite())
    {
      unavailable = true;
      if (wait < 400)   // prevent buffer overflow
        USBSerial.print("USBSerial waiting...");
    }
    wait = 1250;
  }
```
The example does not explicitly invoke flush;
I thought flush was being done in `USB_Send()`.

I downloaded v1.0.1 on 12 Aug.

For my application, I would prefer to know when TX buffers are full or sends are not happening.

Levi--G, 2025-09-08:
> The example does not explicitly invoke flush; I thought flush was being done in `USB_Send()`.

That is correct, this library is supposed to mimic the AVR usb api of arduino, not improve upon it ;) The original implementation also has a busy send that flushes or waits until the transmission is done, the USBCDC I implemented here in fact doesn't use that but instead uses a SendQuick that does not wait because there is no use in waiting for a single write. That is why the flush is implemented as well. In case of a buffer being full however the wait becomes mandatory as there is nowhere for the data to go.

> For my application, I would prefer to know when TX buffers are full or sends are not happening.

Knowing when exactly and which packet a send is not happening is not possible with the buffered approach without flushing each packet. I could add another flush overload that allows you to specify a timeout but this is again further and further from the AVR api that also didn't support this.

You need this buffer because the PC determines when we can send our data, so when your code says "send this" we cannot just dump this data on the usb. We need to store it until the pc requests it, this can take milliseconds up to seconds, or if you close your terminal, indefinitely. I think the main issue here is the availableForWrite is not returning the amount you are expecting to write and i will investigate this later.

Levi--G, 2025-09-10:
My STLink wasn't working today, i'll try again on friday.

Levi--G, 2025-09-13:
Can you check/test the version that is currently on the Delevop branch? I think it might solve your issue, default settings without changes should work fine for your purpose.