Levi--G/USBLibrarySTM32 #12: Problematic MIDI

blekenbleu, 2025-09-14:
## Description of Issue

1) Composite MIDI + CDC fails  
    no valid USB device is seen by Windows PC when both `USBCDC USBserial;` and `MidiUSB.available();`
    - Perhaps my sketch lacks correct code
          https://github.com/blekenbleu/MIDIecho/blob/main/MIDIecho.ino

2) Cannot connect for write MIDI from one app to Black Pill and connect for read MIDI from another app
   - works when only a single app reading and writing MIDI

3) `sendMIDI()` packets work only during `read()`
    explained here: https://github.com/blekenbleu/MIDIecho#midi--hardwareserial

4)  `available()` appears to never return non-zero

## Technical Details

- Arduino Board:  BLACKPILL_F411CE 
- Host OS: Windows 1`
- Arduino IDE / Platformio + Version:  Arduino IDE 2.3.6

## Sketch to reproduce 

~~~
Code/link to code  https://github.com/blekenbleu/MIDIecho
~~~

## Setup Details
Any pin wiring details or other relevant hardware.
- USB TTL serial dongle to GND, PA3, PA2 

## Additional info (optional)
https://github.com/blekenbleu/MIDIecho#midi-echo-example

blekenbleu, 2025-09-15:
> As far as i know this is normal windows/linux behavior, only 1 program can open a device at the same time...

Thanks for the response;  I apologize, this should be a separate enhancement...
While only one app can connect to a single USB MIDI device (and works for my initial application),
 I have several single USB **composite** devices with **separate IN and OUT MIDI devices**, using only 3 endpoints,
 which do in fact allow connecting different apps.
One example is PreSonus AUDIOBOX USB (USB Device Tree Viewer):
```
    =========================== USB Port2 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 1-10-2

      ========================== Summary =========================
Vendor ID                : 0x194F (PreSonus Audio Electronics)
Product ID               : 0x0302
USB Version              : 1.1
Port maximum Speed       : High-Speed
Device maximum Speed     : Full-Speed
Device Connection Speed  : Full-Speed
Self powered             : no
Demanded Current         : 500 mA
Used Endpoints           : 3

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : AudioBox USB
Device Path 1            : \\?\USB#VID_194F&PID_0302#6&a7559d3&0&2#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Device Path 2            : \\?\USB#VID_194F&PID_0302#6&a7559d3&0&2#{cc3df00a-575a-4bfc-8187-1a714939236b}
Kernel Name              : \Device\USBPDO-21
Device ID                : USB\VID_194F&PID_0302\6&A7559D3&0&2
Hardware IDs             : USB\VID_194F&PID_0302&REV_0270 USB\VID_194F&PID_0302
Driver KeyName           : {a20fbcfa-b952-440f-a625-9c06c40e655a}\0000
Driver                   : \SystemRoot\System32\drivers\paeusbaudio.sys (Version: 5.50.0.0  Date: 2023-04-06)
Driver Inf               : C:\WINDOWS\inf\oem83.inf
Legacy BusType           : PNPBus
Class                    : paeusbaudio_sc
Class GUID               : {a20fbcfa-b952-440f-a625-9c06c40e655a}
Service                  : paeusbaudio
Enumerator               : USB
Location Info            : Port_#0002.Hub_#0014
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(10)#USB(2), ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS10)#ACPI(HSS2)
Container ID             : {1ed17275-f064-11ef-97e9-806e6f6e6963}
Manufacturer Info        : PreSonus
Capabilities             : 0x84 (Removable, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : AudioBox USB
  Device Path 1          : \\?\TUSBAUDIO_ENUM#VID_194F&PID_0302&KS#7&36a6ae6f&0&2#{65e8773e-8f56-11d0-a3b9-00a0c9223196}\pcm_out_01_c_00_48000 (AM_KSCATEGORY_RENDER)
  Device Path 2          : \\?\TUSBAUDIO_ENUM#VID_194F&PID_0302&KS#7&36a6ae6f&0&2#{6994ad04-93ef-11d0-a3cc-00a0c9223196}\topology (AM_KSCATEGORY_AUDIO)
  Device Path 3          : \\?\TUSBAUDIO_ENUM#VID_194F&PID_0302&KS#7&36a6ae6f&0&2#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\pcm_in_01_c_00_48000 (AM_KSCATEGORY_CAPTURE)
  Kernel Name            : \Device\000000ad
  Device ID              : TUSBAUDIO_ENUM\VID_194F&PID_0302&KS\7&36A6AE6F&0&2
  Class                  : MEDIA
  Driver KeyName         : {4d36e96c-e325-11ce-bfc1-08002be10318}\0017 (GUID_DEVCLASS_MEDIA)
  Service                : paeusbaudioks
  Location               : Port_#0002.Hub_#0014
   Child Device 1        : Speakers (AudioBox USB) (Audio Endpoint)
    Device Path          : \\?\SWD#MMDEVAPI#{0.0.0.00000000}.{06ac592e-ae4e-4e2b-9275-121624a1689b}#{e6327cad-dcec-4949-ae8a-991e976a79d2} (DEVINTERFACE_AUDIO_RENDER)
    Kernel Name          : \Device\00000102
    Device ID            : SWD\MMDEVAPI\{0.0.0.00000000}.{06AC592E-AE4E-4E2B-9275-121624A1689B}
    Class                : AudioEndpoint
    Driver KeyName       : {c166523c-fe0c-4a94-a586-f1a80cfbbf3e}\0012 (AUDIOENDPOINT_CLASS_UUID)
   Child Device 2        : Line (AudioBox USB) (Audio Endpoint)
    Device Path          : \\?\SWD#MMDEVAPI#{0.0.1.00000000}.{cdd4424d-e91e-4739-82b6-ed538db842b7}#{2eef81be-33fa-4800-9670-1cd474972c3f} (DEVINTERFACE_AUDIO_CAPTURE)
    Kernel Name          : \Device\00000103
    Device ID            : SWD\MMDEVAPI\{0.0.1.00000000}.{CDD4424D-E91E-4739-82B6-ED538DB842B7}
    Class                : AudioEndpoint
[... 423 lines cut ...]
bDescriptorSubtype       : 0x01 (MS General)
bNumEmbMIDIJack          : 0x01 (1 embedded MIDI jack)
baAssocJackID(1)         : 0x03 (Jack(1) ID=3)
Data (HexDump)           : 05 25 01 01 03                                    .%...

      -------------------- String Descriptors -------------------
             ------ String Descriptor 0 ------
bLength                  : 0x04 (4 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language ID[0]           : 0x0409 (English - United States)
Data (HexDump)           : 04 03 09 04                                       ....
             ------ String Descriptor 1 ------
bLength                  : 0x1E (30 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "PreSonus Audio"
Data (HexDump)           : 1E 03 50 00 72 00 65 00 53 00 6F 00 6E 00 75 00   ..P.r.e.S.o.n.u.
                           73 00 20 00 41 00 75 00 64 00 69 00 6F 00         s. .A.u.d.i.o.
             ------ String Descriptor 2 ------
bLength                  : 0x1A (26 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "AudioBox USB"
Data (HexDump)           : 1A 03 41 00 75 00 64 00 69 00 6F 00 42 00 6F 00   ..A.u.d.i.o.B.o.
                           78 00 20 00 55 00 53 00 42 00                     x. .U.S.B.

```

Levi--G, 2025-09-15:
> Thanks for the response; I apologize, this should be a separate enhancement... While only one app can connect to a single USB MIDI device (and works for my initial application), I have several [...]

Yes? I fail to see what you expect the USB stack to do differently?

blekenbleu, 2025-09-15:
> what you expect the USB stack to do differently

define a **USB composite device** with separate `MIDI OUT` and `MIDI IN` devices..?

Levi--G, 2025-09-16:
> Do you use a version of STM32 other than 2.11.0?

I use latest STM32, it updates automatically for me.
 
> As it happens, this is my first experience with Arduino MIDI for Pluggable USB; (having previously used only TinyUSB with ESP32).

No problem

> FWIW, USBMIDI _also_ claims to work with `PluggableUSB` https://docs.arduino.cc/libraries/usbmidi/

Yes but i suggest choosing one and not mixing them since that can give issues. USBMIDI is also just not tested since i have never used it.

> FWIW, sketch for composite USB device with a pair of CDC instances also hangs at `while (!USB_Running())` https://github.com/blekenbleu/SerialEcho

Are you still doing this in arduino IDE? Have you tried adjusting the USB_CFGBUFFER_LEN as i asked before?
Atm with these options:
	-D USB_CFGBUFFER_LEN=256
	-D USB_EP_SIZE=32
I am running a blue pill with 2 USBCDC and a MIDIUSB instance

[image]

Levi--G, 2025-09-17:
> Yes, i have `#define USB_CFGBUFFER_LEN 256`
> 
> > -D USB_EP_SIZE=32
> 
> Oops, my sketch reports 64; I changed that in `USBOptions.h` to 32, but it still hangs at `while (!USB_Running())`..
> 
> A comment there seems relevant: `// Don't forget to also make the epsize smaller in the interface definition`
> 
> * What/where is that interface definition?

The interface definition is only for custom devices/usb library developers, USBCDC and MIDIUSB handle it correctly and you don't need to change anything there.

At this point I will have to suspect either bad code or the IDE or a bad clone chip since it works on my end with the develop branch and the latest stm32duino version.

Levi--G, 2025-09-17:
> * the Black Pill is a WeAct STM32F411 from a few years ago that wants an A10 jumper for DFU to work

Hold on a second, i thought you had a classic black pill, which is a blue pill with fixed usb resistors? A STM32F411 is a totally different chip with USB_OTG_FS USB module instead of a classic USB module. A USB_FS_USB module actually has 3 endpoints instead of 7 on the USB, 3 endpoints is just not enough for anything other than 1 CDC OR 1 Midi...

>   * CMIS in step 1. immediately baffles me

Why are you using stm32duino develop branch? and why are you trying to set up a stm32duino development environment? I think you might be confused as to what you need here? 😄 

> * Trying to ignore my dislike of Visual Studio Code and ignorance of PlatformIO...
>   
>   * matching all PlatformIO and STM32Duiono options to precisely duplicate a configuration seems implausible

I do not understand what you mean here? Any arduino environment can be easily ported to platformio as it gives more options, not less. 😉

blekenbleu, 2025-09-17:
> Why are you using stm32duino develop branch?

I'm not,  just researching based on your "since it works on my end with the develop branch"

> i thought you had a classic black pill, which is a blue pill with fixed usb resistors

I have several STM32F411 available, but only one STM32F103C8 as yet not deployed.
I just got TTL serial pins installed on that STM32F103C8,
loaded my composite CDC sketch to it...

# it works!

[image]

Levi--G, 2025-09-17:
> > Why are you using stm32duino develop branch?
> 
> I'm not, just researching based on your "since it works on my end with the develop branch"

I meant the develop branch of this library 😄 sorry for the confusion
 
> it works!

Nice, yea i still don't have a solution for the low ep count for 4xx devices, it would also need a custom midi library to work on less ep's so i don't know if its worth the effort to try and fix it...

blekenbleu, 2025-09-17:
> a custom midi library to work on less ep

Thanks for your patience and support.  Life is short;
I've ordered a few more Blue Pills (hoping they are genuine).
I'll try building a Blue Pill sketch with one CDC and 2 MIDI,
allowing separate apps to connect.

Hacking custom MIDI library for separate send and receive devices, saving 2 endpoints,
is much lower priority for me than integrating this.

closed 2025-09-17 by blekenbleu